beacon3d/beacon_klipper #7: FR: Print new z-offset when calling Z_OFFSET_APPLY_PROBE

jmwhelan, 2023-04-27:
When adjusting z-offset during a print, and then issuing a Z_OFFSET_APPLY_PROBE command, it would be good to print out the new z-offset which is how it works with a BLTouch.

This would be good, because I can enter it manually in the printer.cfg instead of waiting for the print to end and call SAVE_CONFIG.

[image]

sboschert, 2023-11-12:
I'd like to have this functionality, too. So I went ahead and created a PR: https://github.com/beacon3d/beacon_klipper/pull/13

dalegaard, 2024-01-29:
Hi,

This was added today in 0a12243.

Best regards,
Lasse

closed 2024-01-29 by dalegaard