beacon3d/beacon_klipper #9: Add support for zero_reference_position Klipper configuration parameter for bed_mesh

YanceyA, 2023-08-06:
The bed_mesh Klipper configuration for zero_reference_position parameter is now depreciated. The Klipper formal documentation now suggests the following: "Existing configurations using the relative_reference_index option must be updated to use the zero_reference_position."

However, zero_reference_position is not supported as a parameter in the Beacon code as confirmed by a Beacon developer via Discord. Can the beacon module be updated to use the zero_reference_position parameter currently recommended by Klipper?

Klipper documentation for these two parameters:
https://www.klipper3d.org/Bed_Mesh.html

sdooweloc, 2023-11-19:
having this updated would be great please. Then we can all remove deprecated code from our configs and make sure we are at parity with Klipper's main branch.

dalegaard, 2024-01-22:
This is supported now! Enjoy!

Best regards,
Lasse

closed 2024-01-22 by dalegaard