beacon3d/beacon_klipper #10: Can not calibrate Z axis

hobbyplace, 2023-10-01:
Hi
After I reinstalled Mainsail I am unable to home Z axis

in clipper console 

BEACON_QUERY
Last reading: 2974276.90Hz, 27.20C, no model

QUERY_ENDSTOPS
x:open y:open z:TRIGGERED

I verified the sensor and the led is off (becomes active if a metal object is detected)

additional info:
Host(aarch64, 32bit)
Version: v0.11.0-279-g7bd32994-dirty
OS: Raspbian GNU/Linux 11 (bullseye)
Distro: MainsailOS 1.2.1 (bullseye) 

mcu beacon(beacon)
Version: Beacon 1.0.0
Load: 0.00, Awake: 0.00, Freq: 32 MHz, 

Do you have any suggestion what else should I check?

nielsvz, 2023-10-03:
I assume this is happening because your config does not have a model defined. Did you run through the getting started guide, for calibrating bacon, specifically? https://docs.beacon3d.com/quickstart/#6-calibrate-beacon

brucetony, 2024-01-02:
I had the same problem when I tried running `G28 X Y`. My problem was that I created a macro to overwrite my `G28` command which would cause it to try and home in the Z direction no matter what. Once I commented out this macro, I could home in the X and Y directions using `G28` again and then run the `BEACON_CALIBRATE` command to create a model to be saved.

closed 2024-01-29 by dalegaard