beacon3d/beacon_klipper #18: Internal error on command:"_BED_MESH_CALIBRATE"

DerDreschner, 2024-02-01:
I use Klipper with adaptive bed mesh before each print. Right now, the print stopped during calibration with an internal error on command "_BED_MESH_CALIBRATE". The logfile for klippy contains the following error message when this happened:

```
Unhandled exception during run
Traceback (most recent call last):
  File "/home/biqu/klipper/klippy/klippy.py", line 217, in run
    self.reactor.run()
  File "/home/biqu/klipper/klippy/reactor.py", line 292, in run
    g_next.switch()
  File "/home/biqu/klipper/klippy/reactor.py", line 340, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/biqu/klipper/klippy/reactor.py", line 158, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/biqu/klipper/klippy/reactor.py", line 48, in invoke
    res = self.callback(eventtime)
  File "/home/biqu/klipper/klippy/extras/beacon.py", line 643, in <lambda>
    self.reactor.register_async_callback(lambda e: self._stream_flush())
  File "/home/biqu/klipper/klippy/extras/beacon.py", line 627, in _stream_flush
    cb(sample)
  File "/home/biqu/klipper/klippy/extras/beacon.py", line 1340, in _handle
    self.cb(sample)
  File "/home/biqu/klipper/klippy/extras/beacon.py", line 1979, in cb
    xi = int(round((x - min_x) / self.step_x))
OverflowError: cannot convert float infinity to integer
Transition to shutdown state: Unhandled exception during run
```

It's the first time something like this happens. My printer.cfg contains the following settings regarding the beacon and bed mesh calibration:

```
[beacon]
serial: /dev/serial/by-id/usb-Beacon_Beacon_RevH_26CEFCBB5154364134202020FF141809-if00
x_offset: -5.22 # update with offset from nozzle on your machine
y_offset: -35.08 # update with offset from nozzle on your machine
mesh_main_direction: y
mesh_runs: 2

[resonance_tester]
accel_chip: beacon
probe_points: 90, 90, 20

...

[bed_mesh]
speed: 175
mesh_min: 15, 45
mesh_max: 285, 255
probe_count: 100, 100
algorithm: bicubic
```

I use a Beacon Rev. H together with a Manta M8P V2.0 board.

cinusik, 2024-02-01:
Same here, going back two version fixed problem. Pleas fix this :)

dalegaard, 2024-02-01:
Hi guys,

Thanks for reporting this. A subtle memory management issue snuck in with a performance optimization in 0a12243. It's been fixed in cb35f9c. Sorry for the inconvenience!

Best regards,
Lasse

closed 2024-02-01 by dalegaard