beacon3d/beacon_klipper #23: Rev.d updated firmware, now fails to connect 6/10 times

SlackHacky, 2024-06-01:
Rev.d updated firmware and software to rev.2 , before when firmware 1 and beacon ver.1 software 100% connection. Now six out of 10 boot fails, no beacon detected on USB - query by get mcu id - via ssh - not there - power down and fully reboot of host and mcu will eventually get a connection. - i have reinstalled from repo , forced flashed firmware 3x ( including deleting revh.dfu, so only revd.dfu there )  Not a  USB cable problem as printer is in a closed space under stairs and USB not unplugged or loose or touched.

vzbot 330 in 2wd but all 4 5160 tmc stepper chips 48v power in
3x tmc2240 on 24v for z1 ,z2 and Extruder

Beacon3d is rev D
Manta m8p v2 and CM4 4gb/16gb emmc on bookworm

usb on manta are usb2 and x2 1 is Beacon3d rev d and other tyro HD webcam - no hub

Kizime123, 2024-06-02:
the beacon team is currently working on a fix for this issue. The best way to keep up with the issue is, https://discord.gg/annexengineering as the github issues aren't tracked.

SlackHacky, 2024-06-03:
NOT fixed - its definitely a not detected via USB on firmware_restart - like i said a full reboot or better still power down and power up, gives a better chance of 'catching' a connection.... will follow on discord - i'm quite happy to downgrade to old v1 firmware if needed or possible.

dalegaard, 2024-06-11:
This was fixed in firmware 2.0.1 If you still have issues, please open a forum thread on Discord.

Best regards,
Lasse

closed 2024-06-11 by dalegaard